SiLab-Bonn/basil #5: Read during write to hardware module register

themperek, 2015-03-11:
If size modulo 8 = 0 no need to read?
https://github.com/SiLab-Bonn/basil/blob/master/host/basil/HL/RegisterHardwareLayer.py#L95

laborleben, 2015-03-11:
where is the error?

themperek, 2015-03-11:
You do not need to read if you write full bytes. You will overwrite full byte whatever.

laborleben, 2015-03-12:
depends on the offset and length. yes, sometimes no read is necessary. Probably it makes sense to catch this particular case.

laborleben, 2015-03-12:
I think this makes the code to complicated to save a few bytes.

closed 2015-03-12 by laborleben

themperek, 2015-03-12:
It is one if and for serial port communication it makes a difference.

laborleben, 2015-03-12:
then I wold fix serial communication (whatever module), and I would not recommend using RegisterHardwareLayer. I don't see any reason to "fix" something that is working fine for 99% of the stuff.
I would use set_data() and get_data() for something that needs to be fast.

themperek, 2015-03-12:
Serial communication (UART) uses HL drivers for modules there is no way around this.