SiLab-Bonn/basil #8: Clean up TL base classes

laborleben, 2015-03-13:
Every layer needs a very simple base class.

laborleben, 2015-03-13:
SiUart also breaks interface

themperek, 2015-03-13:
Does not see how SiUart breaks interface (parameter names should be changed). Last time I have seen at Loe it was written better.

laborleben, 2015-03-13:
different parameter name... exactly. Should be solved. But anyway, a general clean-up is necessary.

laborleben, 2015-03-13:
Leo is probably still working with the SVN ... :-)

themperek, 2015-03-13:
No. He has this concept to check in very 3 mounts.

themperek, 2015-03-13:
Suggest change name SiSerial to Serial and SiVisa to Visa because they do not implement any protocol like SiX.
What you think?

laborleben, 2015-03-13:
full ack

themperek, 2015-03-13:
Should have like SiTransfareLayer as base class for SiX?
Can this API be unified? 
Any ideas? Split something?
I still like that basil give you ability to have name mapping.

laborleben, 2015-03-13:
I recommend using abstract base class (import abc) for this kind of implementation. I can do this in the development branch.

Anyway I'm currently working on reading the calibration ROM for all types of adapter cards. The support for all adapter cards I would like to have in the new release.

DavidLP, 2015-03-14:
I agree with the ideas, but what about renaming the TransferLayer that requires adress, data not SiTransferLayer but something that fits to its task. Since we have multiple end points a fitting word would be MultiplexingTransferLayer (http://en.wikipedia.org/wiki/Transport_layer)

DavidLP, 2015-03-14:
And I guess it is Transport and not Transfer layer

themperek, 2015-05-09:
It looks it is ok now?

closed 2015-06-20 by themperek