SiLab-Bonn/basil #9: remove host folder to make packaging/releasing simpler

laborleben, 2015-03-13:


themperek, 2015-03-13:
I do not know but this probably make sense.

DavidLP, 2015-03-14:
I agree.

themperek, 2015-05-06:
Still makes sense but I would move this to future:
At the moment can be done by: 
```
pip install -e "git+https://github.com/SiLab-Bonn/basil.git#egg=basil&subdirectory=host"
```

laborleben, 2015-05-06:
anything that prevent us to do it now?

themperek, 2015-05-06:
This should be in principle quite safe. Can do symbolic link?
Have to updates howto in all dependencies.

What you think to upload this to pypi? It would need to by under basil-daq or something. basil is taken.

laborleben, 2015-05-06:
the directory is not important if one uses proper imports. If someone still uses sys.path.append("/home/user/workspace/basil/host/") than it is clearly his/her fault.

closed 2015-09-17 by laborleben