SiLab-Bonn/basil #10: Use unit prefix only instead of full unit

laborleben, 2015-03-14:
instead set_voltage(self, channel, value, unit='mV') use set_voltage(self, channel, value, unit='m').

Makes things easier when comes to abstraction. We can also do some unit conversion based on a new class that we can put inside utils. The conversion is then based on a dict like that one:
_prefix = {'y': 1e-24,  # yocto
           'z': 1e-21,  # zepto
           'a': 1e-18,  # atto
           'f': 1e-15,  # femto
           'p': 1e-12,  # pico
           'n': 1e-9,   # nano
           'u': 1e-6,   # micro
           'm': 1e-3,   # mili
           'c': 1e-2,   # centi
           'd': 1e-1,   # deci
           'k': 1e3,    # kilo
           'M': 1e6,    # mega
           'G': 1e9,    # giga
           'T': 1e12,   # tera
           'P': 1e15,   # peta
           'E': 1e18,   # exa
           'Z': 1e21,   # zetta
           'Y': 1e24,   # yotta
    }

laborleben, 2015-03-14:
We can stay compatible to the old interface by just evaluating the first char.

themperek, 2015-03-14:
full ack

DavidLP, 2015-03-14:
I would not add a unit feature at all. It should be understood that all is in SI base units. Adding a e.g. e-3 at the base unit should be no issue. Otherwise one can only understand any value with the unit string. At least I would make this a not mandatory API feature, but an optional way to define values. And the standard unit should be SI base units.

laborleben, 2015-03-14:
depends, actually I can live without that feature. Others are using it.
However, implementing the conversion into every function is a waste of code. It can also be, that a conversion is needed for a DAC or ADC (wich is not using SI base unit).

DavidLP, 2015-03-14:
A conversion for a ADC/DAC? What is that? ADC/DAC is just a value and no unit at all.

themperek, 2015-03-14:
You do not want to thing what is 1.273V in DAC units. And calibration on
top.

DavidLP, 2015-03-14:
Now we are talking about calibration and not units ;-). But for calibration data it is much better to have the SI base units only, otherwise some calibrations will be in mV, some in V... 
To say set_voltage(1.273) is what I would prefer. How this is translated to a DAC value is irrelevant.

themperek, 2015-03-14:
One does not go against other we assume we use default Si unit as default so e-3 etc will work.
And we have optional unit parameter.
Would this work? Does not break current situation.

themperek, 2016-11-10:
http://pint.readthedocs.io/

themperek, 2019-01-07:
We do this?