SiLab-Bonn/basil #11: Clean up HL

DavidLP, 2015-03-14:
Just some ideas for the HL folder for discussion:

1. I would like to clean up the HL base class. It should not have a base adress. This is something only needed for the RegisterHardwareLayer (right?).

2. The interface (self._intf) is an instance of a transfer layer module. So why not to name it self._TL_intf to make this obvious? Or at least we can add a comment that the interface is a TL instance. 

3. Some modules do not use any TL from basil. These modules should be either changed or taken out of basil. They seem to be "stand alone" modules (agilent33250a, iseg_shq) and do require additional modules that are neither shipped with basil nor std. python packages. That makes them pretty "displaced".

4. The naming is ugly. Upper/lower/CapWord convention all mixed up. I would prefer to follow https://www.python.org/dev/peps/pep-0008/#package-and-module-names

5. In total we have description parts (e.g. RegisterHardwareLayer, TransferLayer) and implementations of these (e.g. pulse_gen) in the same folder. I would like to have these seperated, but the question is if we gain something from this. HL looks already crowded now, and the very basic classes are just between their implementations.

6. The scpi device descriptions are different from the other files (yaml file not python script). Maybe better to put it somewhere else? A subfolder? Not shippinh with basil?

themperek, 2015-03-14:
ad. 1. Yes
ad. 2. Long run self._intf can be also any HL driver. Maybe name should be change to _parent or something.
ad. 3. Not sure. They give you name mapping and bit more standard interface.
ad. 4. Yes, BUT modules should have same name as in verilog (and I would stay like this). 
ad. 5. It will get worse. If you find good solution.
ad. 6. Can be but I like name mapping. Can this be done with external package? Maybe some fancy yaml syntax?

DavidLP, 2015-03-14:
3. Both modules can easily use the new Serial / VISA TL, its just not implemented yet.
4. and 5. One could think of restructuring basil in a way that the pure description is in one folder. With all the layers and the virtual description classes. 
And then there is a folder having the concrete implementations (python module + verilog code). Makes it also quite convinient to have theses together in one folder. The seperation now is more like because it is another language not because the purpose is different. But thats just my taste, nothing really needed. Would make the structure more logical to my mind. Question is if this helps users.
6. what is name mapping?

themperek, 2015-03-14:
>    1. Both modules can easily use the new Serial / VISA TL, its just not
>    2. and 5. One could think of restructuring basil in a way that the
>    3. what is name mapping?

ad.3. So one can say dut["VDDA"].set_voltage(1.2) and this will use
multiplier/GPAC or anything else. Maybe not so important.

DavidLP, 2015-03-14:
ad 3. Both (agilent33250a, iseg_shq) are just stand alone implementations to talk to a laboratory device. iseg_shq uses pyiseq hosted on our redmine server and uses pyserial at the end. So it could immediatly use a basil TL. And Agilent33250A cannot be used because the needed module is hosted nowhere (when I am not mistaken). And I think that it has Serial implemented only and not GPIB. And it is a scpi device... It  can use the basil VISA TL immediatly and would profit from this. 
There is no reason (anymore) to make special outside basil TL implementations for simple lab devices.
But at the end I don't care too much... these modules use basil for name mapping only.

laborleben, 2015-03-15:
why should we remove ISEG HL? a driver is available and can be downloaded. I just have to male the documentation better. ISEG HL was also used by Toko.

But this is not the case for Agilent where no interface is available and the new VISA driver can be used.

laborleben, 2015-03-15:
concerning naming conversion:

we used all uppercase for hardware and lowercase for FPGA modules. Exceptions prove the rule.

laborleben, 2015-03-15:
Moving the base classes somewhere else makes it non-obvious what bases classes are already existing.  I would keep them where they are.

DavidLP, 2015-03-15:
One could copy the ISEG HL / Agilent33250A HL implementations into basil and make them use the VISA / Serial TL. The benefit is that the device TL decription would be unified. Thus using them does not depend on the special init APIs (port=COM3, ...) and changing to other devices (like Keithley 2400) and interfaces (Agilent via GPIB) do not require code changes. Also testing of devices using VISA TL via simulation is implemented now. So one could even simulate and unittest the ISEG HL / Agilent33250A HL implementations. And there would be a nice first example of the lab device support ;-).

themperek, 2015-05-06:
@DavidLP  should we close this ?

laborleben, 2015-05-06:
remove agilent33250a.py from HL? A VISA TL driver is available. 

Iseg_shq.py I would like to keep for the time being, though I agree with David to switch to Serial TL in the future.

themperek, 2015-05-06:
agree

laborleben, 2015-05-06:
I will close this thread. Please open new tickets on specific topics.

closed 2015-05-06 by laborleben

DavidLP, 2015-05-07:
agilent33250a.py is based on Tokos implementation (thus adding unit conversion) and is already based on VISA TL. So one could keep it for Toko, or as an example how to derive from scpi class.

laborleben, 2015-05-07:
ok restored.