SiLab-Bonn/basil #21: TDC module with inverted trigger and tigger delay does not work

DavidLP, 2015-04-29:
Only a positive edge trigger for the trigger delay determination is working (= delay between edges of TDC and trigger) . If a negative edge trigger is used it fails.

# Not working

TLU cfg:
EN_INVERT_TRIGGER = 1
TDC cfg:
ENABLE = True
EN_TRIGGER_DIST = True
[image]

# Working

TLU cfg:
EN_INVERT_TRIGGER = 0
TDC cfg:
ENABLE = True
EN_TRIGGER_DIST = True
[image]

laborleben, 2015-04-29:
Use TDC invert trigger.

closed 2015-04-29 by laborleben

DavidLP, 2015-04-30:
More info:
The concrete implementation of the TDC / TLU module in pyBAR (where this was tested with) is such, that the trigger signal is routed through the TDC module. Thus is you want to accept inverted triggers, it has to be set in the TDC module and not in the TLU module:

self.dut['tdc_rx2']['EN_INVERT_TRIGGER'] = 1

As an enhancement the trigger signal routed through the TDC module should not be affected by EN_INVERT_TRIGGER to make the configuration of these modules independet from each other.

laborleben, 2015-04-30:
this is also fixed now, the TDC outputs are not affected anymore