SiLab-Bonn/basil #24: usb3 and SiLibUsb

themperek, 2015-05-06:
Finally should have usb3 in main branch

There are few possibilities. 
One suggestion can be to meke new TL called SiUsb3 (at the end this can look better then some ugly if). If not one need to add error message in SiUsb if wrong version of SiLibUsb is used.

Nevertheless one need to change SiLibUsb api or count count argument for FastBlockWrite and FastBlockRead or just always use WriteExternal and ReadExternal only.

What you think?

ViacheslavFilimonov, 2015-05-07:
I think that adding a new SiUsb3 TL is a good idea. However, in this case user should care about it (in case of "if" he shouldn't).

Yes, we should merge SiLibUsb apis.

themperek, 2015-05-07:
One nevertheless needs different yaml file for usb3 so this is probably no issue.

In SiLibUsb you can implement in a way that looks how many arguments is given to  WriteExternal/ReadExternal if 2 then address is 0 if 3 then use also address?
Check both *args and **kwargs. Or just always use 3. @laborleben

laborleben, 2015-05-07:
after all the pros and cons I would also prefer a SiUSB3 TL. This will make the source code less obscure.

laborleben, 2015-05-09:
This goes into the new release?

themperek, 2015-05-09:
As we discussed this should be 5-10min work. The most work in not basil related.

laborleben, 2015-05-09:
Yep, so nothing to stop us from making a new release.

themperek, 2015-05-09:
One in general in future should probably make smaller incremental releases. Not like this one.
@laborleben  correct this: https://github.com/SiLab-Bonn/basil/releases/tag/untagged-06177df272610ec7ea17

laborleben, 2015-07-23:
I cherry picked SiUsb3 TL. It is now in development.

closed 2015-08-19 by themperek