SiLab-Bonn/basil #47: Autoupload to PyPI

DavidLP, 2016-02-19:
Shall i add autoupload to PyPI for successfull release builds?

laborleben, 2016-02-19:
Yes, sure. Both sdist and bdist_wheel would be great.

DavidLP, 2016-06-14:
Mhh I think it does not work, does it?

laborleben, 2016-06-14:
Did not last time.

DavidLP, 2016-06-14:
True. The log on Travis is unfortunately inconclusive. We could always upload to pypitest for all development comits to check this.

DavidLP, 2016-06-20:
I uploaded the standard package format for the 2.4.3 release. Wheel only does not seem to be sufficient, since Linux operating systems cannot use it, even if it's non binary ?!.

laborleben, 2016-06-20:
cannot use a source package?!? That is weird...

DavidLP, 2016-06-20:
Yes, that was not possible.

themperek, 2017-12-14:
@DavidLP This is done?

laborleben, 2018-11-03:
Still failing:
```
PyPI upload failed.
failed to deploy
```

laborleben, 2018-11-03:
Maybe https://pypi.org/help/#verified-email?

laborleben, 2020-03-12:
Fixed by commit 0b00265113cc92560f9d91a7e7cce0fe9b7dd44b.

closed 2020-03-12 by laborleben