SiLab-Bonn/basil #48: Setting power gives error

themperek, 2016-02-19:
@laborleben 
'''
    self.dut['ADAPTER_CARD'].set_voltage('VDDD1', 1.2)
  File "/basil/basil/HL/FEI4AdapterCard.py", line 161, in set_voltage
    self._set_dac_value(value=value, **self._ch_map[channel]['DACV'])
  File "/basil/basil/HL/FEI4AdapterCard.py", line 85, in _set_dac_value
    self._intf.write(self._base_addr + self.MAX_520_ADD, array('B', pack('BB', channel, value)))
error: ubyte format requires 0 <= number <= 255
'''

laborleben, 2016-02-19:
This is known, some adapter cards do not reach 1.2V. It might also be that the calibration EEPROM is broken or is not set at all.

themperek, 2016-02-19:
For this module I cannot see lower then 1.2. For 1.2 I get DAC raw value of 272 which is > 255 (allowed).  It works with 1.3.
Should it log a warning and set 255?

laborleben, 2016-02-19:
yeah right a log warning should be printed. try: ...set_dac_value() except Error: ...logging.warning()

But be careful: sometimes an error is also given when setting to 0V (also outside the range of the DAC). Simply setting the DAC to maximum (e.g. 255) is not a good idea in this case.

themperek, 2016-02-19:
One can also set default to 1.25V or something and this goes away. Should not meter for the chip operation.

laborleben, 2016-02-19:
IMHO not touching the DAC and not setting the DAC to something else is the best solution. I would keep it as it is and implement a nicer error message. The user has to decide what to do.