SiLab-Bonn/basil #53: Configuration yaml not working

michaeldaas, 2016-05-23:
When a device is initialized like this:

```
 transfer_layer:
  - name  : device name
    type  : device type
    additional_parameter : Bar
```
Then the additional parameter(s) are not taken into account. This bug (or feature, why?) was introduced in this commit: 6f9091a

themperek, 2016-05-23:
Full example please.

michaeldaas, 2016-05-23:
The qMCA software as is works if i revert to an older version and does not, when i delete the three lines you deleted in the commit above...

themperek, 2016-05-23:
Where?

DavidLP, 2016-05-23:
Look at your commit 6f9091a. If you do not know what this did and what it does, I would recomment to revert.

laborleben, 2016-05-23:
use
```
transfer_layer:
  - name  : device name
    type  : device type
    init:
        additional_parameter : Bar
```

DavidLP, 2016-05-23:
I guess it has to be under init now

michaeldaas, 2016-05-23:
It works now with the additional parameters under init, thanks.

laborleben, 2016-05-23:
keep it as it is, and use basil as it is intended. We removed the code by intention, we kept the backward compatibility for several years. We decided to remove that.

DavidLP, 2016-05-23:
Config file format changed. Parameter have to be under init.

closed 2016-05-23 by DavidLP