SiLab-Bonn/basil #61: test of SCPI fails

laborleben, 2016-08-28:
ERROR: test_write (test_SimScpi.TestSimScpi)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/SiLab-Bonn/basil/tests/test_SimScpi.py", line 43, in test_write
    self.device['Pulser'].set_on()
  File "/home/travis/build/SiLab-Bonn/basil/basil/HL/scpi.py", line 65, in method
    raise ValueError('Invalid SCPI command %s for device %s' % (name, self.name))
ValueError: Invalid SCPI command set_on for device Pulser

The error in the test never occurred because set_on command was never executed because "if .. elif .." statement had no final "else". I added "else" recently, so the error is visible now.

What was the intention of "set_on" / "get_on"? These commands doesn't exist in our device drivers. There is only "on" / "off". We might just implement "set_on" / "get_on" / "set_off" / "get_off" if that makes sense.

DavidLP, 2016-08-29:
It is a simulated device and the commands [do exist](https://github.com/SiLab-Bonn/basil/blob/development/basil/HL/scpi_sim_device.yaml).

DavidLP, 2016-08-29:
Is working again after [this](https://github.com/SiLab-Bonn/basil/commit/ac9fe543b6e9564a6f3403cbd400330958329fe2) commit.

closed 2016-08-29 by DavidLP

laborleben, 2016-08-29:
Thanks, do you know where pyvisa-py has the @sim backend definition?

DavidLP, 2016-08-29:
Pyvisa-sim it is a special backend for pyvisa for testing if no hardware is available. Documentation is [here](http://pyvisa-sim.readthedocs.io/en/latest/). In BASIL [this line](https://github.com/SiLab-Bonn/basil/blob/development/tests/test_SimScpi.py#L21) tells pyvisa to use the simulation backend.

laborleben, 2016-08-29:
I know, but I see only "?IDN" support for standard "@sim" backend. Where are the these commands defined:
identifier : ERROR
set_on : "!OUT"
get_on : "?OUT"
get_frequency : "?FREQ"

Additional commands (like the ones above) can be implemented via a yaml file. But I cant find that.

laborleben, 2016-08-29:
OK, it is an extra package ... pip install -U pyvisa-sim

DavidLP, 2016-08-29:
Correct.

laborleben, 2016-08-29:
OK, the default device is [here](https://github.com/hgrecco/pyvisa-sim/blob/master/pyvisa-sim/default.yaml):

But ?IDN should return "LSG Serial #1234". But the identifier in [this](https://github.com/SiLab-Bonn/basil/blob/development/basil/HL/scpi_sim_device.yaml) file is "ERROR".

print(inst.query("?IDN"))
LSG Serial #1234

laborleben, 2016-08-29:
Default IEEE 488.2 commands are in a dict.

DavidLP, 2016-08-29:
Yep, that's nicer