SiLab-Bonn/basil #63: Better support for other oscilloscopes (e.g. Tektronix MSO4034)

laborleben, 2016-08-28:
Retrieving waveforms from various oscilloscope may vary from device to device.
Therefore a better support for retrieving the raw data is necessary.
For example, the driver for the MSO4104b works also for the MSO4034 but retrieving raw data is different.
For the MSO4034 the following settings are necessary:
resource.query_binary_values("DATA:SOURCE CH1;:WAVFrm?", datatype='b', is_big_endian=True)

laborleben, 2016-08-29:
one can achieve even higher resolution when using 2 byte readout (e.g. AVErage mode uses up to 14 bits per point):
r.write("DATa:WIDth 2") # 2 byte per value
r.write("DATa:ENCdg RIBINARY") # signed integer
r.query_binary_values("DATA:SOURCE CH1;:WAVFrm?", datatype='h', is_big_endian=True)

laborleben, 2016-08-29:
I found the following steps in the Tektronix Programmer Manual:
"
You can transfer waveforms from the oscilloscope to an external controller using the following sequence:
1. Select the waveform source using the DATa:SOUrce command.
2. Specify the waveform data format using DATa:ENCdg.
3. Specify the number of bytes per data point using DATa:WIDth.
4. Specify the portion of the waveform that you want to transfer using DATa:STARt and DATa:STOP.
5. Transfer waveform preamble information using the WFMPRe? query.
6. Transfer waveform data from the oscilloscope using the CURVe? query.
"

DavidLP, 2016-08-29:
What do you mean with better support? You can call the methods as you mentioned them.
Generally to use a python package that helps us parsing raw data would be a nice addition to
the SCPI class.

laborleben, 2016-08-29:
Right now every oscilloscope may have different default settings. Also reading the data via DATA:SOURCE CHX;:WAVFrm? just breaks things because data and preamble is mixed in the same data stream and makes interpretation more complicated.
First fix would be to split up preamble and data (see point 5 and 6).

DavidLP, 2016-08-29:
What do you mean with _just breaks things_? We are using the TEKTRONIX MSO4104B successfully in pyBar via BASIL. I agree data taking can be done faster in binary mode, but this is not something we need right now? In BASIL only TEKTRONIX MSO4104B is supported, so the issue title should be _Support for other oscilloscopes_ and not better support. Because there is none. And if we want to support other Oszis (or pulsers, etc.) then we have to find a solution how to parse raw data from different SCPI devices in a general way.

laborleben, 2016-08-29:
Yes it is needed right now, CERN has only MSO4034 :-)

DavidLP, 2016-08-29:
Ok, I see :-). Just copy and past the device driver that is already there and change the lines for retreiving the waveform.