SiLab-Bonn/basil #75: SiTCP: underlying object is terminated before readout thread is terminated.

laborleben, 2017-08-25:
The thread is sometimes terminated after the underlying SiTCP object is terminated resulting in incomplete errors messages:
```
Exception in thread TcpReadoutThread (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/home/jens/miniconda2/lib/python2.7/threading.py", line 801, in __bootstrap_inner
```
or 
```
Exception in thread TcpReadoutThread (most likely raised during interpreter shutdown):
Traceback (most recent call last):
```
or
```
Exception in thread TcpReadoutThread (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/home/jens/miniconda2/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/home/jens/miniconda2/lib/python2.7/threading.py", line 754, in run
```

The reason is an AttributeError because the attribute doesn't exist anymore.
The implementation of connect() and disconnect() can solve that issue.

themperek, 2017-08-25:
This: https://stackoverflow.com/questions/24512846/python-how-to-stop-a-thread-thats-waiting-for-a-recv ?

laborleben, 2017-08-29:
One of the answers is suggesting to let it die. But this is not satisfactory. The second part of this answer is suggesting to use the Event class. This also requires some action by the main program (i.e. calling disconnect()/close() function).
The best protection against crashes of the main application is to add a context manager to the Dut class. If proper implemented, this may close any open connection (USB, Socket, etc.) in case of troubles.

The code may look similar to this:
```Python
with Dut(conf.yaml) as dut:
    dut.init()
```

themperek, 2017-08-29:
Maybe just use ```def __del__(slef)``` close by event and wait for it to close?

laborleben, 2017-08-29:
well ```__del__()``` is not guaranteed to be executed. This answer in this thread https://stackoverflow.com/questions/1481488/what-is-the-del-method-how-to-call-it pretty much summarizes it all.

michaeldaas, 2018-01-05:
I would like to bump this issue, because we could use a Dut.close() method that terminates the thread and closes the socket in bdaq53. Is there any chance you could implement this?