SiLab-Bonn/basil #76: StdRegister offset

themperek, 2017-12-14:
As pointed by @hansk68  `offset` should point to LSB and not to MSB position. 
Changing this will break a lot of compatibilities.

Options:
- break compatibility
- remove `offset` and change this something else like `lsb_pos` used here: https://www.vmmcentral.org/uvm_vmm_ik/files3/reg/uvm_reg_field-svh.html
- add new filed and deprecate `offset`

Comments welcome.

laborleben, 2017-12-14:
This is an old discussion...

We use ```offset``` in ```HL.RegisterHardwareLayer``` in the sense of LSB position.

I'd recommend to stick with "offset" and fix the behavior. You can implement error checks for the offset and throw an error with a meaningful error message.

themperek, 2017-12-15:
For v2.5.0? It may lead to compatibility issues at the begining.

laborleben, 2017-12-16:
So, do we add a new class for that?

themperek, 2017-12-16:
This seems ok. To FieldRegister or just Register?

laborleben, 2017-12-18:
I vote for Register

themperek, 2019-01-07:
I started to work on this for 3.0.

There is one complication. At the moment, `repeat ` in `StdRegister` put element 0 on left (MSB) side of the array and I think to be consistent it should be on the right (LSB) of the array so I am planning to change this in new `Register` class too.
Any opinion on this @laborleben @thirono @MarcoVogt

laborleben, 2019-01-08:
I agree.