SiLab-Bonn/basil #83: Logging for bronkhorst_elflow instead of pring

themperek, 2018-05-13:
@thirono Could you change `print` to `logger` here:
https://github.com/SiLab-Bonn/basil/blob/development/basil/HL/bronkhorst_elflow.py

Example: https://github.com/SiLab-Bonn/basil/blob/development/basil/HL/NTCRegister.py

`+` format the source

DavidLP, 2018-05-14:
Also I recommend to put a link to the protocol description and/or a few lines about the protocol. See for example how it is done in basil for other hardware [here](https://github.com/SiLab-Bonn/basil/blob/development/basil/HL/sensirion_ekh4.py#L15) or [here](https://github.com/SiLab-Bonn/basil/blob/development/basil/HL/weiss_sb22.py#L12).

leloup314, 2023-02-18:
@AntonioT7 you could solve this issue in your fork, would be good chance to practice ;)

cbespin, 2024-05-28:
I am closing this as I do not see any `print` statements in the module anymore (whenever they got fixed).

closed 2024-05-28 by cbespin