SiLab-Bonn/basil #85: TCP timeout

YannickDieter, 2018-06-14:
With the latest version of basil (development) I got last testbeam sometimes this kind of error:

```console
Traceback (most recent call last):
 File "meta_tune_threshold_simple.py", line 193, in <module>
   mtt.start(**local_configuration)
 File "meta_tune_threshold_simple.py", line 46, in start
   self.scan(**kwargs)
 File "meta_tune_threshold_simple.py", line 139, in scan
   th1, scurve, mean, tdac1 = correct(1, tdac)
 File "meta_tune_threshold_simple.py", line 89, in correct
   scan.start(**kwargs)
 File "/home/silab/git/bdaq53/bdaq53/scan_base.py", line 473, in start
   self.chip.init()
 File "/home/silab/git/bdaq53/bdaq53/rd53a.py", line 335, in init
   super(RD53A, self).init()
 File "/home/silab/git/basil/basil/dut.py", line 119, in init
   catch_exception_on_init(item)
 File "/home/silab/git/basil/basil/dut.py", line 113, in catch_exception_on_init
   mod.init()
 File "/home/silab/git/basil/basil/TL/SiTcp.py", line 175, in init
   self._sock_tcp.connect((self._init['ip'], self._init['tcp_port']))
 File "/home/silab/miniconda2/lib/python2.7/socket.py", line 228, in meth
   return getattr(self._sock,name)(*args)
socket.timeout: timed out
```

Before I never observed this and after switching one commit behind this, the error was never again observed during testbeam. Is it maybe related to this commit 99f77a300d8b3acf2a5dcef94ff9a4acc9cd470e?

I tested this change before merging once with pymosa and no issue was observed.

YannickDieter, 2018-06-14:
Okay is it [this](https://github.com/SiLab-Bonn/basil/blob/development/basil/TL/SiTcp.py#L160) parameter?

laborleben, 2018-06-14:
yes, but this can be changed in the yaml file as well.

YannickDieter, 2018-06-14:
Okay, I will try this.

YannickDieter, 2018-06-14:
After setting the timeout to 5 seconds, the error does not pop up again. Should we change the default again to 5 seconds?

laborleben, 2018-06-14:
Yes, please.

YannickDieter, 2018-06-15:
Default is now set to 5 seconds, which is a better default. I will close now this issue.

closed 2018-06-15 by YannickDieter