SiLab-Bonn/basil #91: Clock domain crossing issues in m26_rx

laborleben, 2018-08-28:
Reading the timestamp uses the clock from an external source where the timestamp itself is synchronous to an internal clock.

See: https://github.com/SiLab-Bonn/basil/blob/07b7cf30d4607caa5df677098d9207d0ec0b1c61/firmware/modules/m26_rx/m26_rx_core.v#L149

themperek, 2018-08-28:
The user has to know what is dong and use CLK_RX to generate the timestamp (or any other clock in proper time domain).

Actually issue is in pymosa here: https://github.com/SiLab-Bonn/pymosa/blob/master/firmware/src/mmc3_m26_eth.v#L558-L570 (this is not done correctly) but this is not basil issue

laborleben, 2018-08-28:
Yes, with the gray coded timestamp there are no CDC issues. I wasn't aware of the fact.

closed 2018-08-28 by laborleben