SiLab-Bonn/basil #108: default_nettype

prymaszewski, 2019-03-18:
**Problem:**
A macro from Xilinx's IP catalog gave a nettype errors during synthesis in Vivado 2018.2, complaining about nettype, even though the exact same code was fine in a example design generated by Vivado. Error message example:

`[Synth 8-3966] non-net port TXCLKLOCKED_I cannot be of mode input: default_nettype is "none" ["/faust/user/prymaszewski/git/RD53CDRBONN/daq/firmware/vivado/cdr53bdaq.srcs/sources_1/ip/ibert_7series_gtx_0/hdl/ibert_7series_gtx_v3_0_syn_rfs.v":152]`

**Cause:**
Some of the modules in basil/firmware/modules/utils have a `default_nettype none` declaration (e.g. [clock_divider.v](https://github.com/SiLab-Bonn/basil/blob/99052482d9334dd1f5598eb2d2fb4d5399a32291/basil/firmware/modules/utils/clock_divider.v#L8), [fifo_32_to_8.v](https://github.com/SiLab-Bonn/basil/blob/99052482d9334dd1f5598eb2d2fb4d5399a32291/basil/firmware/modules/utils/fifo_32_to_8.v#L8) ).

**Solution for my case:**
Add `default_nettype wire` in the top file after includes of modules from basil/firmware/modules/utils (has to be after includes, otherwise default_nettype is changed to none).

**Question:**
Is the `default_nettype none` declaration useful? Could it be removed or changed to `default_nettype wire`?

themperek, 2019-03-18:
`default_nettype none` is useful and can save you from many problems. Xilinx does not code IP properly. Can be overwritten by setting `default_nettype wire` in the Xilinx code or before include.

closed 2019-03-18 by themperek

prymaszewski, 2019-03-18:
Fine, but in case someone experiences the same problem - the `default_nettype wire` has to be set after the includes, not before, otherwise it does not work.

Modyfing Xilinx's code is not an option sometime e.g. in my case the IBERT code is encrypted.