SiLab-Bonn/basil #109: module with multiple classes outside of basil

thirono, 2019-03-25:
If the modules has multiple classes, it cannot be imported

themperek, 2019-03-25:
example?

thirono, 2019-03-25:
This will work:

themperek, 2019-03-25:
Propose PR?

thirono, 2019-03-25:
but this will not work

`hw_drivers:`
`  - name      : GPAC
`    type      : monopix_daq.customizedGPAC
`    interface : intf
`    base_addr : 0x00000
`    init :
`        no_calibration : True

laborleben, 2019-03-25:
But now you change several parameters at once. Which item is the one you are referring to?

Be careful when changing the import function. Please check this commit https://github.com/SiLab-Bonn/basil/commit/41a2b6a68fa44850f464b18e0d30d1f5260b84e0 and the related issue: https://github.com/SiLab-Bonn/basil/issues/92

thirono, 2019-03-25:
never mind, it was already fixed in new basil

closed 2019-03-25 by thirono