SiLab-Bonn/basil #121: Measuring SMU current/voltage resets current/voltage limit

flooklab, 2019-10-17:
The `get_current()`/`get_voltage()` functions for Keithley 2400 SMU (maybe other models as well?) send the commands `MEAS:CURR?` / `MEAS:VOLT?`. This does not only measure something but also resets the previously set current/voltage limit to its default value again.

```
>>> dut['SMU'].set_current_limit(1e-6)
>>> dut['SMU'].set_voltage(-2)
>>> dut['SMU'].on()
>>> 
>>> dut['SMU'].get_current_limit()
'1.000000E-06\r'
>>> 
>>> dut['SMU'].get_current()
'-2.000000E+00,-1.470096E-08,+9.910000E+37,+6.860023E+03,+2.151000E+04\r'
>>> 
>>> dut['SMU'].get_current_limit()
'1.050000E-04\r'

```

Is this the expected behavior? Should we maybe change this or at least add an additional function that measures while leaving the limits unchanged?
There exists a `READ?` command, which returns the measured values in the same format
while not changing the limits.

See also https://gitlab.cern.ch/silab/bdaq53/issues/301

laborleben, 2019-10-17:
This is certainly not a desired behavior. Can you please check the Keithley 2400 user manual and give us more information and verify with your device? Thanks.

FlorianHinterkeuser, 2019-10-17:
According to the user manual, the `MEAS[:FUNC]?` is essentially a `READ?`: it first configures the SMU for the following read (to measure either volts, amps or ohms. Can also be done manually by `CONF[:FUNC]`) and then sends a `READ?`. It should not under any circumstances reset the source/measurement range. 
The other devices of this family (2410, 2000 Multimeter, to some extend 26xx as they use the same commands) do not show this behaviour. One should check with another 2400 SMU. If all 2400 have this feature, we should use `READ?` instead and need to be careful to configure the device before reading.

michaeldaas, 2019-10-18:
according to Keithley themselves, "The MEAS? query resets the unit and other things."...

see https://forum.tek.com/viewtopic.php?t=95780

flooklab, 2019-10-18:
Checked with another 2400. The newer one is not affected.

michaeldaas, 2019-10-18:
mhh. we could actually check the version of the device from the init string and use the correct function accordingly..

DavidLP, 2019-12-17:
Please somebody check, who has a Keithley 2400

flooklab, 2019-12-18:
#125 works fine with Keithley 2400.

DavidLP, 2019-12-19:
Great, thx.

commit 4a219164 (2019-12-19): BUG: DEP: try fix #121
closed 2019-12-19 by DavidLP