SiLab-Bonn/basil #166: Q: Possibility to add formatter to device description yaml

leloup314, 2022-01-11:
When using `basil` to read some quantity e.g. current off of an SMU, the resulting return string format depends on the SMU model. This is inconvenient when using different SMU models, even though their API through `basil` is always the same e.g `smu.get_current()`. In my measurement script I end up with something like this:

    if device_type == 'keithley_2410':
        return float(result.split(',')[1])

    elif device_type == 'keithley_6517A':
        return float(result.split(',')[0][-4])
    ...

Is it possible to add something like a `formatter` to the device description yaml from which the the return value of `get_some_quantity` is formatted in such a way that the resulting string contains just the respective quantity's value? e.g.
    
    # Keithley 2410
    smu.get_current()  # Returns '1.23'
    
    # Keithley 6517A
    smu.get_current()  # Returns '1.23'

DavidLP, 2022-01-11:
I always wanted to add a parser feature that a format string of return values can be defined in a generic way. Since I did not find a suitable python parser package quickly I skipped this endeavor. One could define how to parse the return value in the yaml file

DavidLP, 2022-01-11:
Maybe there is something like this already there, without defining your own generic parser

leloup314, 2022-01-11:
Maybe some `regex` would be useful

leloup314, 2022-02-09:
This has been merged. Closing

closed 2022-02-09 by leloup314