SiLab-Bonn/basil #175: Q: Initialization procedure

leloup314, 2022-03-07:
Is there a particular reason for the following Syntax to be necessary?

    from basil.dut import Dut
    
    # Instantiate my_device
    my_device = Dut(my_config)
    my_device.init()  # Additionally call an init method on the instance manually

Why not call `my_device.init` inside the  instances `__init__`?

themperek, 2022-03-08:
I do not remember anymore but there was a reason for this. 
We sometimes manipulate things before actually connecting to hardware.

leloup314, 2022-03-08:
From my subjective perspective it looks like 90% of the use cases do not need to do this.
Maybe it would be nice to have something like a default argument `Dut(auto_init=True)` that automatically calls the `init` as last action inside the `__init__` if set. If not wanted, set `auto_init=False`

themperek, 2022-03-08:
>Dut(auto_init=False)

This should be OK.

leloup314, 2022-03-08:
But of course this would break some stuff. My point is to design the API for the edge-cases is not the right approach. Can you tell me where you do something in between the initailization of the Dut and the init call?

themperek, 2022-03-08:
In general OK but not sure if this is a good idea to do now. Should accumulate more braking changes?

themperek, 2022-03-08:
Maybe not best example: https://gitlab.cern.ch/silab/bdaq53/-/blob/development/bdaq53/system/bdaq53.py

leloup314, 2022-03-08:
The only line where something happens in between initialization and manual call to `init` is for the [clock generator](https://gitlab.cern.ch/silab/bdaq53/-/blob/development/bdaq53/system/bdaq53.py#L116). Everything else looks fine to me.

leloup314, 2022-03-08:
But yes, maybe we can wait and have a more thorough look at this, maybe accumulate other braking changes.

cbespin, 2022-03-10:
> Maybe not best example: https://gitlab.cern.ch/silab/bdaq53/-/blob/development/bdaq53/system/bdaq53.py

Just to add, since [tj-monopix2-daq](https://github.com/SiLab-Bonn/tj-monopix2-daq) is almost exactly like bdaq, this will also be affected. No problem for me to adapt, though.