SiLab-Bonn/basil #237: Mercury controller board number misused as device (address) number

Giakou, 2025-02-20:
The board number that is read back with the `TB` command is misused as the controller address in the Mercury class. As you can see below in the screenshots from the manual of the controller `board_number = controller_address - 1`. 

[image]
[image]

I don't know how heavily this class is used among projects, but there are two solutions here:

1. Replace `address` with `board_number` to reflect the manual instructions. This will not break things, but the user SHOULD do the calculation in their head -> `board_number = controller_address - 1`.
2. Fix the class so it really uses the actual hardware `address` you set with the switches. However, this will break running systems.

What do you think @cbespin , @YannickDieter , @MarcoVogt ?

cbespin, 2025-02-20:
If address selection uses the board number anyway, I would be in favour to use this one instead of the address. If you interpret the DIP switches as binary, one starts counting at 0 anyway. So the address information seems useless to me, or am I missing something?
I think the problem is the wording in the manual, that adds 1 to the board number and labels it `address`.

Giakou, 2025-02-20:
I agree that the wording in the manual is strange. However, I think that there should be 1:1 connection of the software to the manual and no assumptions should be made. Even if it's not intuitive what the manual calls "address" the software needs to adapt the same definitions. 

I think replacing "address" by "board_number" or something similar is a better solution as well.

rpartzsch, 2025-02-20:
I had the same problem in [motor_stage_ui](https://github.com/SiLab-Bonn/motor_stage_ui). There, I just shift the address value by one when communicating with the controller. So I don't use the `board_number` explicitly. I think if you just open the manual and check how to set addresses on the controller, this is the way one would understand how to operate the motor stage.

cbespin, 2025-02-20:
I have not really used the driver, so if the concerned people come to a common solution, I am happy. But just for my understanding:

https://github.com/SiLab-Bonn/basil/blob/4b86048c2721ebaecb42a51340a1e68bc75886f2/basil/HL/mercury.py#L52-L54

The `address` parameter that is given to the function is actually from a `range(16)`, so would go from 0 to 15. And the function returns the `board_number` which is actually `address - 1`. Which means when calling `get_address(0)` we will never get a return value and we can never call `get_address(16)` which should return 15 (which is the board number when all 4 DIP switches in the same position).

And this board number is actually what is passed to all the methods of the class? Would it be fine to rename the method `get_board_number()`, call it with values from 1 to 16 in the `init()` and replace all occurrences of `address` with `board_number`. The manual and the class call it "TB" only, so I think this would be good enough to stay true to the manual. And this way, one would not have to check code in existing implementations.

cbespin, 2025-02-21:
I never used this module, so from my side, you can just propose something, that ideally does not break functionality of existing implementations. Or check with other users. We can ask in Wednesday's meeting who actually uses the module and check with them.
Otherwise, just propose something and I will probably merge it :D

PR #238 (2025-03-26): Use board number instead of address in the Mercury class
Fixes #237